schnitzeltony/meta-qt5-extra #18: k18n: bad installation prefix

plfiorini, 2016-11-23:
I'm putting together a distro that uses meta-qt5-extra, when building ki18n I have this error:

```
WARNING: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets.cmake contains links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include!
WARNING: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets.cmake contains links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include!
WARNING: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets-noconfig.cmake contains links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib!
ERROR: One or more files in sysroot contain links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr
ERROR: Function failed: do_sysroot_cmake_sanity (log file is located at /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/temp/log.do_populate_sysroot.177)
ERROR: Logfile of failure stored in: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/temp/log.do_populate_sysroot.177
ERROR: Task 6161 (/home/plfiorini/git/liri/meta-liri/sources/meta-qt5-extra/recipes-kde/kf5/tier1/ki18n/ki18n-native.bb, do_populate_sysroot) failed with exit code '1'
```

The mentioned `log.do_populate_sysroot.177` log:

```
DEBUG: Executing python function sstate_task_prefunc
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing python function do_populate_sysroot
DEBUG: Executing shell function sysroot_stage_all
0 blocks
0 blocks
grep: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/staged_cmake_files: No such file or directory
DEBUG: Shell function sysroot_stage_all finished
DEBUG: Executing python function sysroot_strip
DEBUG: runstrip: 'strip' --remove-section=.comment --remove-section=.note --strip-unneeded '/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/libKF5I18n.so.5.27.0'
DEBUG: Python function sysroot_strip finished
DEBUG: Executing python function relocatable_binaries_preprocess
DEBUG: Python function relocatable_binaries_preprocess finished
DEBUG: Python function do_populate_sysroot finished
DEBUG: Executing python function do_qa_staging
NOTE: QA checking staging
DEBUG: Python function do_qa_staging finished
DEBUG: Executing shell function do_sysroot_cmake_sanity
WARNING: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets.cmake contains links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include!
WARNING: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets.cmake contains links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include!
WARNING: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets-noconfig.cmake contains links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib!
ERROR: One or more files in sysroot contain links to /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr
ERROR: Function failed: do_sysroot_cmake_sanity (log file is located at /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/temp/log.do_populate_sysroot.177)
```

`run.do_configure` reveals this configure command:

```
        cmake \
          ${OECMAKE_SITEFILE} \
          /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/ki18n-5.27.0 \
          -DCMAKE_INSTALL_PREFIX:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr \
          -DCMAKE_INSTALL_BINDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/bin \
          -DCMAKE_INSTALL_SBINDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/sbin \
          -DCMAKE_INSTALL_LIBEXECDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/ki18n \
          -DCMAKE_INSTALL_SYSCONFDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/etc \
          -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/com \
          -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/var \
          -DCMAKE_INSTALL_LIBDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib \
          -DCMAKE_INSTALL_INCLUDEDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include \
          -DCMAKE_INSTALL_DATAROOTDIR:PATH=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/share \
          -DCMAKE_INSTALL_SO_NO_EXE=0 \
          -DCMAKE_TOOLCHAIN_FILE=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/toolchain.cmake \
          -DCMAKE_VERBOSE_MAKEFILE=1 \
               -DOE_QMAKE_PATH_PREFIX=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr     -DOE_QMAKE_PATH_HEADERS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include/qt5     -DOE_QMAKE_PATH_LIBS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib     -DOE_QMAKE_PATH_ARCHDATA=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib     -DOE_QMAKE_PATH_DATA=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/share     -DOE_QMAKE_PATH_BINS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/bin     -DOE_QMAKE_PATH_LIBEXECS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/libexec     -DOE_QMAKE_PATH_PLUGINS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/plugins     -DOE_QMAKE_PATH_IMPORTS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/imports     -DOE_QMAKE_PATH_QML=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/qml     -DOE_QMAKE_PATH_TRANSLATIONS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/share/translations     -DOE_QMAKE_PATH_DOCS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/share/doc     -DOE_QMAKE_PATH_SETTINGS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/etc     -DOE_QMAKE_PATH_EXAMPLES=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/share/examples     -DOE_QMAKE_PATH_TESTS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/share/tests     -DOE_QMAKE_PATH_HOST_PREFIX=     -DOE_QMAKE_PATH_HOST_BINS=/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/bin     -DOE_QMAKE_PATH_HOST_DATA=/home/plfiorini/git/liri/meta-liri/build-inte [... cut ...] /kuitsetup.h
-- Installing: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/image/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/include/KF5/KI18n/ki18n_export.h
-- Installing: /home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/image/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/mkspecs/modules/qt_KI18n.pri
```

plfiorini, 2016-11-23:
Indeed `/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/work/x86_64-linux/ki18n-native/5.27.0-r0/sysroot-destdir//home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr/lib/cmake/KF5I18n/KF5I18nTargets.cmake` has `_IMPORT_PREFIX` set to `/home/plfiorini/git/liri/meta-liri/build-intel-corei7-64/tmp/sysroots/x86_64-linux/usr` probably it should be `/usr/

schnitzeltony, 2016-11-23:
Strange: Had never had this in my latest builds. Do you have

commit c03b32bd71dbe04f2f239556fea0b53215e403d7
Author: Thomas Witt <Thomas.Witt@bmw.de>
Date:   Mon Aug 22 16:27:11 2016 +0200

    cmake.bbclass: call cmake with a relative path

    CMake wants a relative path for CMAKE_INSTALL_*DIR, an absolute path
    breaks cross-compilation. This fact is documented in the following
    ticket: https://cmake.org/Bug/view.php?id=14367

    $sysconfdir and $localstatedir are not relative to $prefix, so they are
    still set as absolute paths. With his change ${PROJECT}Targets.cmake
    that are generated by cmakes "export" function will contain relative
    paths instead of absolute ones.

    Signed-off-by: Thomas Witt <Thomas.Witt@bmw.de>
    Signed-off-by: Clemens Lang <clemens.lang@bmw-carit.de>
    Signed-off-by: Ross Burton <ross.burton@intel.com>

in your oe-core?

plfiorini, 2016-11-23:
I don't think I have this commit. git log of poky/meta/classes/cmake.bbclass doesn't show it.
The distro uses poky jethro rather than plain oe.
Looks like poky krogoth has it: http://git.yoctoproject.org/cgit/cgit.cgi/poky/log/meta/classes/cmake.bbclass?h=krogoth

schnitzeltony, 2016-11-23:
> cgit/cgit.cgi/poky/log/meta/classes/cmake.bbclass?h=krogoth

The mentioned commit is necessary - it caused huge rework. Lots of the
cmake-lib path alignments could be removed. Maybe you want to try jethro+
branch of meta-qt5-extra - that should work but does not have latest
versions.

plfiorini, 2016-11-24:
@schnitzeltony Thanks looks like that was indeed the problem. The krogoth build moves further.

closed 2016-11-24 by plfiorini