schnitzeltony/meta-qt5-extra #22: knotifications rdepends on libxtst

plfiorini, 2017-01-05:
```
WARNING: knotifications-5.29.0-r0 do_package_qa: QA Issue: knotifications rdepends on libxtst, but it isn't a build dependency, missing libxtst in DEPENDS or PACKAGECONFIG? [build-deps]
```

schnitzeltony, 2017-01-23:
Cannot reproduce this here. Do you have x11 in your DISTRO_FEATURES?

plfiorini, 2017-01-23:
Yep: https://github.com/lirios/meta-liri/blob/master/conf/distro/lirios.conf#L53

schnitzeltony, 2017-01-23:
I get the feeling I am missing QA messages due to using (standard meta-qt5) /usr/lib/qt5. You have set different path (guess that from #24) - correct?

plfiorini, 2017-01-23:
I have a different path due to meta-boot2qt, see my comment on #24

schnitzeltony, 2017-02-17:
Fixed by 9aa3bb299faa5e61448985639facb04149ccb1ff

closed 2017-02-17 by schnitzeltony