schnitzeltony/meta-qt5-extra #26: kauth fails to build with last change to polkit-qt-1

plfiorini, 2017-02-17:
After d831f1c853241fbd71a84c2b0698a2a5e8741935 I have issues building kauth (machine raspberrypi2 with vc4graphics):

```
| CMake Error at /home/plfiorini/git/liri/meta-liri/build-raspberrypi2/tmp/sysroots/raspberrypi2/usr/lib/cmake/PolkitQt5-1/PolkitQt5-1Targets.cmake:82 (message):
|   The imported target "PolkitQt5-1::Core" references the file
| 
|      "/home/plfiorini/git/liri/meta-liri/build-raspberrypi2/tmp/sysroots-components/cortexa7hf-neon-vfpv4/polkit-qt-1/usr/lib/libpolkit-qt5-core-1.so.1.112.0"
| 
|   but this file does not exist.  Possible reasons include:
| 
|   * The file was deleted, renamed, or moved to another location.
| 
|   * An install or uninstall procedure did not complete successfully.
| 
|   * The installation package was faulty and contained
| 
|      "/home/plfiorini/git/liri/meta-liri/build-raspberrypi2/tmp/sysroots/raspberrypi2/usr/lib/cmake/PolkitQt5-1/PolkitQt5-1Targets.cmake"
| 
|   but not all the files it references.
| 
| Call Stack (most recent call first):
|   /home/plfiorini/git/liri/meta-liri/build-raspberrypi2/tmp/sysroots/raspberrypi2/usr/lib/cmake/PolkitQt5-1/PolkitQt5-1Config.cmake:86 (include)
|   src/ConfigureChecks.cmake:96 (find_package)
|   src/CMakeLists.txt:5 (include)
```

As a matter of fact the lib doesn't exist. Reverting the commit fixes the build.

BTW I'm using morty now, I had the same issue with krogoth.

schnitzeltony, 2017-02-17:
Yes it was the start making meta-qt5-extra build with oe-core master's recipe specific sysroot change. I found out that I need a lot of patience for this transition and aborted these actions. Polkit-Qt patch was reverted. Can you pull and try a build please?

plfiorini, 2017-02-19:
@schnitzeltony Works thanks!

closed 2017-02-19 by plfiorini

wzyy2, 2017-04-22:
Hi,

I try to build it with the latest oe-core and this error is back.
After repick  Polkit-Qt patch,  the build is fixed.

wzyy2, 2017-04-22:
ok.. i understand it wrong.. This commit is for lastest oe-core.