schnitzeltony/meta-qt5-extra #33: Yocto packagegroup-lxqt-base Failed tasks kwidgetsaddons do_compile

denyste, 2017-11-04:
Hi,
I am building an image with Yocto Project (morty). The layer "meta-qt5-extra" includes the recipe "packagegroup-lxqt-base" in my project. However the building generated the following failed tasks:

/*
2 errors

oe_runmake failed

Function failed: do_compile (log file is located at /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/temp/log.do_compile.20270)
*/

Why "kwidgetsaddons" generated these errors and how do I resolve this issue? 

See below the build log for more details: 

NOTE: Logging error 2: {'pathname': '/home/user-test/poky-morty/_toaster_clones/_git___git.yoctoproject.org_poky_morty/bitbake/lib/bb/__init__.py', 'lineno': 100, 'build': <Build: 105 Project object distro-with-lxqt>, 'message': 'oe_runmake failed', 'level': 2}
ERROR: oe_runmake failed
NOTE: Logging error 2: {'pathname': '/home/user-test/poky-morty/_toaster_clones/_git___git.yoctoproject.org_poky_morty/bitbake/lib/bb/build.py', 'lineno': 589, 'build': <Build: 105 Project object distro-with-lxqt>, 'message': 'Function failed: do_compile (log file is located at /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/temp/log.do_compile.20270)', 'level': 2}
ERROR: Function failed: do_compile (log file is located at /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/temp/log.do_compile.20270)
NOTE: recipe kwidgetsaddons-5.30.0-r0: task do_compile: Failed
ERROR: Task (/home/user-test/poky-morty/_toaster_clones/_git___github.com_schnitzeltony_meta-qt5-extra.git_morty/recipes-kde/kf5/tier1/kwidgetsaddons/kwidgetsaddons.bb:do_compile) failed with exit code '1'
NOTE: Logfile for task /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/polkit-qt-1/0.112.0-r0/temp/log.do_populate_sysroot.22244
NOTE: Logfile for task /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/polkit-qt-1/0.112.0-r0/temp/log.do_populate_sysroot.22244
NOTE: Logfile for task /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kguiaddons/5.30.0-r0/temp/log.do_configure.22142
NOTE: Tasks Summary: Attempted 2115 tasks of which 2110 didn't need to be rerun and 1 failed.
NOTE: No package dependencies created
NOTE: No package dependencies created
WARNING: KeyError in save_target_file_information'dirs' 

Would you please help me resolve them?
I thank you in advance.

schnitzeltony, 2017-11-04:
On Sat, Nov 4, 2017 at 11:16 AM, denyste <notifications@github.com> wrote:

> Hi,
> I am building an image with Yocto Project (morty). The layer
> "meta-qt5-extra" includes the recipe "packagegroup-lxqt-base" in my
> project. However the building generated the following failed tasks:
>
> /*
> 2 errors
>
> oe_runmake failed
>
> Function failed: do_compile (log file is located at
> /home/user-test/poky-morty/build-toaster-2/tmp/work/
> cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/
> 5.30.0-r0/temp/log.do_compile.20270)
> */
>
> Why "kwidgetsaddons" generated these errors and how do I resolve this
> issue?
>
> See below the build log for more details:
>
> NOTE: Logging error 2: {'pathname': '/home/user-test/poky-morty/_
> toaster_clones/_git___git.yoctoproject.org_poky_morty/bitbake/lib/bb/
> *init*.py', 'lineno': 100, 'build': <Build: 105 Project object
> distro-with-lxqt>, 'message': 'oe_runmake failed', 'level': 2}
> ERROR: oe_runmake failed
> NOTE: Logging error 2: {'pathname': '/home/user-test/poky-morty/_
> toaster_clones/_git___git.yoctoproject.org_poky_morty/bitbake/lib/bb/build.py',
> 'lineno': 589, 'build': <Build: 105 Project object distro-with-lxqt>,
> 'message': 'Function failed: do_compile (log file is located at
> /home/user-test/poky-morty/build-toaster-2/tmp/work/
> cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/
> 5.30.0-r0/temp/log.do_compile.20270)', 'level': 2}
> ERROR: Function failed: do_compile (log file is located at
> /home/user-test/poky-morty/build-toaster-2/tmp/work/
> cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/
> 5.30.0-r0/temp/log.do_compile.20270)
> NOTE: recipe kwidgetsaddons-5.30.0-r0: task do_compile: Failed
> ERROR: Task (/home/user-test/poky-morty/_toaster_clones/_git___github.
> com_schnitzeltony_meta-qt5-extra.git_morty/recipes-kde/
> kf5/tier1/kwidgetsaddons/kwidgetsaddons.bb:do_compile) failed with exit
> code '1'
> NOTE: Logfile for task /home/user-test/poky-morty/
> build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-
> linux-gnueabi/polkit-qt-1/0.112.0-r0/temp/log.do_populate_sysroot.22244
> NOTE: Logfile for task /home/user-test/poky-morty/
> build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-
> linux-gnueabi/polkit-qt-1/0.112.0-r0/temp/log.do_populate_sysroot.22244
> NOTE: Logfile for task /home/user-test/poky-morty/
> build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-
> linux-gnueabi/kguiaddons/5.30.0-r0/temp/log.do_configure.22142
> NOTE: Tasks Summary: Attempted 2115 tasks of which 2110 didn't need to be
> rerun and 1 failed.
> NOTE: No package dependencies created
> NOTE: No package dependencies created
> WARNING: KeyError in save_target_file_information'dirs'
>
> Would you please help me resolve them?
> I thank you in advance.
>
> Honestly - I am not using morty and it would cause lots of efforts for me
to check. Whatever:

A look into

/home/user-test/poky-morty/build-toaster-2/tmp/work/
cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/
5.30.0-r0/temp/log.do_compile

gives more information what goes wrong with kwidgetsaddons.

denyste, 2017-11-04:
I do not know why "kwidgetsaddons" has failed.
I think the problem comes with "kwidgetsaddons" during the compiling. 
What recipes "kwidgetsaddons" depend on? 
I have looked carefully the "packagegroup-lxqt-base" but why is it not there?
I do not have more information about this error.
I only know that layers and recipes are working with OpenEmbedded. 
The Yocto project uses OpenEmbedded to build an image.
There is two ways to build an image, one with "bitbake" (terminal) and the other with Toaster (web interface)
I only used Toaster.

schnitzeltony, 2017-11-04:
You have further information. The first log you sent says:

> ERROR: Function failed: do_compile (log file is located at /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/temp/log.do_compile.20270)

And I suggested to look into that. Now I suggest to upload it here so I can look into.

Another suggestion: Could you try to build with bitbake to see if the error still occurs?

denyste, 2017-11-06:
These two-way "user interfaces" (bitbake with terminal and Toaster with browser) are different paths and configuration files. 
For example: /home/user-test/poky-morty/build/conf/local.conf, etc. and  /home/user-test/poky-morty/build-toaster-2/conf/toaster.conf, etc.  
There are currently no configuration for "bitbake" (default settings), but the configuration files for "Toaster" are updated.

Please see below the last lines error code of the file log.do_compile.20270:

/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qlist.h:382:13:   required from 'QList<T>& QList<T>::operator<<(const T&) [with T = QVariant]'
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qvariant.h:765:27:   required from here
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qlist.h:448:39: warning: cast from 'QList<QVariant>::Node*' to 'QVariant*' increases required alignment of target type [-Wcast-align]
     else if (QTypeInfo<T>::isComplex) reinterpret_cast<T*>(n)->~T();
                                       ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qshareddata.h:46:0,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qsharedpointer.h:45,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtGui/qpixmap.h:47,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtGui/qbrush.h:51,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtGui/qpalette.h:45,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtWidgets/qwidget.h:47,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtWidgets/qcombobox.h:43,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtWidgets/QComboBox:1,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/kwidgetsaddons-5.30.0/src/kdatecombobox.h:25,
                 from /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/kwidgetsaddons-5.30.0/src/kdatecombobox.cpp:20:
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qhash.h: In instantiation of 'static QHash<K, V>::Node* QHash<K, V>::concrete(QHashData::Node*) [with Key = QString; T = QVariant; QHash<K, V>::Node = QHashNode<QString, QVariant>]':
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qhash.h:533:13:   required from 'static void QHash<K, V>::deleteNode2(QHashData::Node*) [with Key = QString; T = QVariant]'
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qhash.h:573:5:   required from 'void QHash<K, V>::freeData(QHashData*) [with Key = QString; T = QVariant]'
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qhash.h:252:45:   required from 'QHash<K, V>::~QHash() [with Key = QString; T = QVariant]'
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qvariant.h:779:30:   required from here
/home/user-test/poky-morty/build-toaster-2/tmp/sysroots/cubietruck/usr/include/qt5/QtCore/qhash.h:235:16: warning: cast from 'QHashData::Node*' to 'QHash<QString, QVariant>::Node* {aka QHashNode<QString, QVariant>*}' increases required alignment of target type [-Wcast-align]
         return reinterpret_cast<Node *>(node);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: Leaving directory '/home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/build'
CMakeFiles/Makefile2:3914: recipe for target 'src/CMakeFiles/KF5WidgetsAddons.dir/all' failed
make[1]: *** [src/CMakeFiles/KF5WidgetsAddons.dir/all] Error 2
make[1]: Leaving directory '/home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/build'
Makefile:141: recipe for target 'all' failed
make: *** [all] Error 2
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/user-test/poky-morty/build-toaster-2/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/kwidgetsaddons/5.30.0-r0/temp/log.do_compile.20270)

denyste, 2017-11-06:
Please find attached file [log.do_compile.zip]
[log.do_compile.zip](https://github.com/schnitzeltony/meta-qt5-extra/files/1446908/log.do_compile.zip)

All packages with the prefix "Qt" are version 5.7.1+git0+a55f36211e in my project.

schnitzeltony, 2017-11-06:
Ahh - OK symptom is clear - will check later at home how to solve...

schnitzeltony, 2017-11-06:
I have created a morty-next branch and hope that fixes your problem - it would take ages for me to test as I am in the middle of something else...
Be aware that it rebuilds a lot - qtbase was changed and let me know if the fix works for you.

denyste, 2017-11-08:
There appears to be a new issue with the morty-next branch.
After the parsing,  the "tasks starting" are frozen (see the attached screenshot in jpg format).
[X-lxqtshot-2017-11-08_10-21-23.zip](https://github.com/schnitzeltony/meta-qt5-extra/files/1453630/X-lxqtshot-2017-11-08_10-21-23.zip)

I have tried to build several times, but nothing happens.

schnitzeltony, 2017-11-08:
To support there is more information required - log-files particularly - the screenshot tells me nothing.

denyste, 2017-11-08:
Please refer to the following files: 
"truncated_toaster_ui.log" from "toaster_ui.log" because the file was too large (13 MBytes), 
[truncated_toaster_ui.zip](https://github.com/schnitzeltony/meta-qt5-extra/files/1454792/truncated_toaster_ui.zip)

and the file from /home/user-test/poky-morty/build-toaster-2/cache,
[local_file_checksum_cache.zip](https://github.com/schnitzeltony/meta-qt5-extra/files/1454795/local_file_checksum_cache.zip)

and copy and paste content from the terminal,
[copy_content_terminal.zip](https://github.com/schnitzeltony/meta-qt5-extra/files/1455007/copy_content_terminal.zip)

I hope that all of this will help you.

schnitzeltony, 2017-11-08:
Most helpful information I found in truncated_toaster_ui.zip. Something times out. Since I never saw this kind of error and never used Toaster (and never will) I cannot help you on this.
I suggest you ask at yocto mailing list [1]. Tell your problem / the layers you added / add a link to this thread / mention that you use debian-9.1 (builder complained for that). I will follow the thread and if it turns out that it is meta-qt5-extra problem (or other layer I am involved) I try to fix.

For now I close this issue.

[1] https://lists.yoctoproject.org/listinfo/yocto

closed 2017-11-08 by schnitzeltony

denyste, 2017-11-08:
Ok ! I thank you for your support and suggestion.
Maybe later I will use the useful "bitbake" commands.