schnitzeltony/meta-qt5-extra #36: redclaration issue with libepoxy headers and GLES2

prakash56755, 2017-12-07:
In file included from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/qt5/QtGui/qopengl.h:109:0,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/qt5/QtGui/5.9.0/QtGui/qpa/qplatformopenglcontext.h:59,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/work/aarch64-poky-linux/kwin/5.9.1-r0/kwin-5.9.1/plugins/qpa/abstractplatformcontext.h:25,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/work/aarch64-poky-linux/kwin/5.9.1-r0/kwin-5.9.1/plugins/qpa/abstractplatformcontext.cpp:20:
| /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/GLES2/gl2.h:513:90: error: 'void epoxy_glDrawArrays(GLenum, GLint, GLsizei)' redeclared as different kind of symbol
|  GL_APICALL void         GL_APIENTRY glDrawArrays (GLenum mode, GLint first, GLsizei count);
|                                                                                           ^
| In file included from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/epoxy/gl.h:94:0,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/epoxy/egl_generated.h:9,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/epoxy/egl.h:46,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/work/aarch64-poky-linux/kwin/5.9.1-r0/kwin-5.9.1/plugins/qpa/abstractplatformcontext.h:23,
|                  from /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/work/aarch64-poky-linux/kwin/5.9.1-r0/kwin-5.9.1/plugins/qpa/abstractplatformcontext.cpp:20:
| /home/prakash/tickets/anritusu_yocto/aniritusu_bak/anirt/build-kde-xway/tmp/sysroots/zcu102-zynqmp/usr/include/epoxy/gl_generated.h:10479:49: note: previous declaration 'void (* epoxy_glDrawArrays)(GLenum, GLint, GLsizei)'

prakash56755, 2017-12-07:
Any way to solve this?

prakash56755, 2017-12-07:
libepoxy version --> 1.3.1, for morty branch

schnitzeltony, 2017-12-12:
I've seen your issue but haven't found time for it. Please put all related information here and please do not contact me directly in the future.

In your email you wrote:

> When I disabled the qt5 opengl ES2 PACKAGECONFIG[gles2], I am not getting the redeclaration error.

My question: What in your meta-universe enables that? If I am not mistaken meta-qt5 and meta-qt5-extra do not enable PACKAGECONFIG[gles2] by default.

Xou are using meta-xilinx as BSP (correct?). What is the preferred acceleration for your board GL or GLES? If there is no acceleration for your board available, using KDE is no fun anyway...

prakash56755, 2017-12-12:
Sorry for the inconvenience, Thanks for the prompt reply, meta-xilinx layer enables the eglfs,gles2 in qt5, Our board uses the libmali gles acceleration.

Also I found the macro in kwin pacakge in the file kwinglutils_funcs.h.

// qopengl.h declares GLdouble as a typedef of float when Qt is built
// with GLES support.  This conflicts with the epoxy/gl_generated.h
// declaration, so we have to prevent the Qt header from being #included.

#define QOPENGL_H

#ifndef QOPENGLF_APIENTRY
#define QOPENGLF_APIENTRY GLAPIENTRY
#endif

#ifndef QOPENGLF_APIENTRYP
#define QOPENGLF_APIENTRYP GLAPIENTRYP
#endif

if you aware about the above line, can you explain what the above macro will do ?

Thanks in Advance!!

schnitzeltony, 2017-12-12:
Honestly
1. the code sequence does not tell me much yet
2. Have no time to take care till weekend: This layer is a hobby project to me to me and I am under fire for 'real-life' issues.

Till then: are you forced to use morty for a specific reason? Have you tried (or are you willing to try) meta-qt5-extra/master + rocko? Maybe later version of libepoxy behaves better...

prakash56755, 2017-12-13:
Our product tied with morty branch, so we are specifically targetting morty branch. We will try the rocko branch with qt5-extra layer. 

Thank you very much for your kind support

joern477, 2018-02-27:
Are there any news about this issue?
Currently I'm hitting the same problem during build for raspberry pi.
- yocto rocko
- libepoxy 1.4.3

schnitzeltony, 2018-02-27:
Not really - your environment this sounds more or less close to my current environment and I am fine here with

Build Configuration:
BB_VERSION        = "1.37.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "fedora-27"
TARGET_SYS        = "arm-angstrom-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "angstrom"
DISTRO_VERSION    = "v2018.02"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta-angstrom     = "work-2017-12:d99865607d3410033e5ed07f88c42f05a32a9a4e"
meta-oe           
meta-filesystems  
meta-gnome        
meta-initramfs    
meta-multimedia   
meta-networking   
meta-python       
meta-webserver    
meta-xfce         = "work-2018-03:70cda762c6eb8c9df731f362509f20133dee805c"
meta-browser      = "work-2017-12:751ab36eaeafe1c37551dc0be4ae5d30555ac2bc"
meta-raspi-light  = "master:e806f862b8adf7c3c485f0f87d96e1b7778018b1"
meta-gumstix-community = "master:db5f84ff73372472a60b75ea1849745d1691f321"
meta-bbone        = "master:9462ee4d636fdb0c7de64eef3998a3419f53c256"
meta-misc         = "master:3612fef28871bb009ea7ddc0c8f48c89516da2e7"
meta-office       = "master:3213948bafb7fae4fa00ce6dc997d1801e870b5e"
meta-qt5          = "rocko:db881bb4917628ad3eb1fa7c4f27915cf2ae60f4"
meta-qt5-extra    = "master:38691112e6cc98e978e176e36b16031783aebd5a"
meta-games        = "work-2017-12:b7e4f39548c9dea35f481928da79b3f82628c5f6"
meta-retro        = "master:07916c7dc7cdd114dc72acd210b4fac515b92f5c"
meta              = "work-2017-12:ebdf57cf7b9090433a4c735b1465daf698de85a0"

The 'work..' layers I keep here at github and they are sometimes rebased. Maybe your configuration might help.

joern477, 2018-03-01:
The qtbase configuration in use below (shortened version).
Is your Desktop OpenGL enabled or disabled?

Qt Gui:
...
  EGL .................................... yes
  OpenVG ................................. no
  OpenGL:
    **Desktop OpenGL ....................... no**
    OpenGL ES 2.0 ........................ yes
    OpenGL ES 3.0 ........................ no
    OpenGL ES 3.1 ........................ no
  Session Management ..................... yes
...
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    EGLFS EGLDevice ...................... no
    EGLFS GBM ............................ no
    EGLFS Mali ........................... no
    EGLFS Raspberry Pi ................... yes
    EGL on X11 ........................... no
  LinuxFB ................................ yes
  VNC .................................... yes
  Mir client ............................. no
  X11:
    Using system-provided XCB libraries .. yes
    EGL on X11 ........................... no
    Xinput2 .............................. yes
    XCB XKB .............................. yes
    XLib ................................. yes
    XCB render ........................... yes
    XCB GLX .............................. yes
    XCB Xlib ............................. yes
    Using system-provided xkbcommon ...... yes

schnitzeltony, 2018-03-01:
I have (with rocko Qt 5.9)

Building on: linux-oe-g++ (arm, CPU features: neon)
Building for: linux-oe-g++ (arm, CPU features: neon)
Configuration: cross_compile use_gold_linker compile_examples enable_new_dtags largefile neon silent shared release c++11 c++14 c++1z concurrent dbus reduce_exports stl
Build options:
  Mode ................................... release
  Optimize release build for size ........ no
  Building shared libraries .............. yes
  Using C++ standard ..................... C++1z
  Using ccache ........................... no
  Using gold linker ...................... yes
  Using new DTAGS ........................ yes
  Using precompiled headers .............. no
  Using LTCG ............................. no
  Target compiler supports:
    NEON ................................. yes
  Build parts ............................ examples libs tests tools
Qt modules and options:
  Qt Concurrent .......................... yes
  Qt D-Bus ............................... yes
  Qt D-Bus directly linked to libdbus .... yes
  Qt Gui ................................. yes
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. yes
  Qt Widgets ............................. yes
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  QML debugging .......................... yes
  udev ................................... yes
  Using system zlib ...................... yes
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ no
  GLib ................................... yes
  iconv .................................. no
  ICU .................................... yes
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  Using system PCRE2 ..................... no
Qt Network:
  getaddrinfo() .......................... yes
  getifaddrs() ........................... yes
  IPv6 ifname ............................ yes
  libproxy ............................... no
  OpenSSL ................................ yes
    Qt directly linked to OpenSSL ........ no
  SCTP ................................... no
  Use system proxies ..................... yes
Qt Gui:
  Accessibility .......................... yes
  FreeType ............................... yes
    Using system FreeType ................ yes
  HarfBuzz ............................... yes
    Using system HarfBuzz ................ yes
  Fontconfig ............................. yes
  Image formats:
    GIF .................................. yes
    ICO .................................. no
    JPEG ................................. yes
      Using system libjpeg ............... yes
    PNG .................................. yes
      Using system libpng ................ yes
  EGL .................................... yes
  OpenVG ................................. no
  OpenGL:
    Desktop OpenGL ....................... no
    OpenGL ES 2.0 ........................ yes
    OpenGL ES 3.0 ........................ yes
    OpenGL ES 3.1 ........................ yes
  Session Management ..................... yes
Features used by QPA backends:
  evdev .................................. yes
  libinput ............................... no
  INTEGRITY HID .......................... no
  mtdev .................................. no
  tslib .................................. no
  xkbcommon-evdev ........................ yes
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    EGLFS EGLDevice ...................... yes
    EGLFS GBM ............................ yes
    EGLFS Mali ........................... no
    EGLFS Raspberry Pi ................... no
    EGL on X11 ........................... yes
  LinuxFB ................................ no
  VNC .................................... yes
  Mir client ............................. no
  X11:
    Using system-provided XCB libraries .. yes
    EGL on X11 ........................... yes
    Xinput2 .............................. yes
    XCB XKB .............................. yes
    XLib ................................. yes
    XCB render ........................... yes
    XCB GLX .............................. yes
    XCB Xlib ............................. yes
    Using system-provided xkbcommon ...... yes
Qt Widgets:
  GTK+ ................................... yes
  Styles ................................. Fusion Windows
Qt PrintSupport:
  CUPS ................................... no
Qt Sql:
  DB2 (IBM) .............................. no
  InterBase .............................. no
  MySql .................................. no
  OCI (Oracle) ........................... no
  ODBC ................................... no
  PostgreSQL ............................. no
  SQLite2 ................................ no
  SQLite ................................. yes
    Using system provided SQLite ......... yes
  TDS (Sybase) ........................... no

Fyi: Have additional packageconfigs set in [1-2] maybe there is something which needs migration to meta-qt5-extra (unlikely) - or meta-raspberrypi.

[1] https://github.com/schnitzeltony/meta-misc/blob/master/appends-packageconfig/meta-qt5/qtbase_%25.bbappend
[2] https://github.com/schnitzeltony/meta-raspi-light/blob/master/appends-qt5/qtbase/qtbase_%25.bbappend

schnitzeltony, 2018-03-01:
One question: You said you are using meta-raspberrypi. Do you have VC4 (full mesa) enabled?

joern477, 2018-03-01:
I've not actively switched to VC4, so I would guess I use the proprietary driver.

One side note, I'm not using 
`meta-qt5-extra`
I'm using:
`https://phabricator.kde.org/source/yocto-meta-kf5/`
`https://phabricator.kde.org/source/yocto-meta-kde/`

Only the issue discussed here seems to be the same. Sorry, forgot to mention that ;-)
Maybe the configurations are more different then I thought. 

I see you already apply some patches to [kwin](https://github.com/schnitzeltony/meta-qt5-extra/tree/master/recipes-kde/plasma/untier/kwin)

On KDE's Phabricator the issue discussed here is also tracked: https://phabricator.kde.org/D10762

schnitzeltony, 2018-03-01:
> I'm using:
> https://phabricator.kde.org/source/yocto-meta-kf5/
> https://phabricator.kde.org/source/yocto-meta-kde/

Wow - I am maintaining this shit for almost 4 years now and have to admit it was useless efforts - have to remove KDE soon it is just a waste of time!!!

schnitzeltony, 2018-03-01:
I am not going to answer KDE related questions anymore - the know better

closed 2018-03-01 by schnitzeltony