schnitzeltony/meta-qt5-extra #51: Kwin does not see fontconfig

megastallman, 2018-07-15:
Hi @schnitzeltony ! Thanks for the great project, so I'm trying to get it built by little steps.
Here is the unsuccessful one: https://hastebin.com/jatazewivo.sql

Shortly, I've got fontconfig built already, just checking that it is built, but when I run "bitbake kwin" it says:
```
|   Could NOT find Fontconfig (missing: FONTCONFIG_LIBRARIES
|   FONTCONFIG_INCLUDE_DIR)
| Call Stack (most recent call first):
|   /home/metaldoctor/yb-kde-1/tmp/work/core2-64-poky-linux/kwin/5.13.2-r0/recipe-sysroot-native/usr/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
|   cmake/modules/FindFontconfig.cmake:45 (find_package_handle_standard_args)
|   CMakeLists.txt:256 (find_package)
```
Thanks in advance!

schnitzeltony, 2018-08-03:
Sorry for late reply - was offline for a while.

1. What does 'bitbake -e kwin | grep ^DISTRO_FEATURES' report? 
2. Have you tried to add fontconfig to kwin's DEPENDS

megastallman, 2018-08-03:
Thanks! Could you please add fontconfig to Kwin's depends too? It fixed my problem.

closed 2018-08-03 by megastallman

schnitzeltony, 2018-08-03:
Done - 7935e5d4851e2df91ee8121d2a6159424446b949