schnitzeltony/meta-qt5-extra #52: Conflict between sip and sip3

willdlcre, 2018-08-02:
Hi everyone, 

I'm trying to build an image including the python3-pyqt5 package. However, during the do_rootfs task of my image recipe's build, I got this error : 

```Collected errors:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - nothing provides sip3 = 4.19.2-r0 needed by sip3-dev-4.19.2-r0.cortexa9hf-neon
 * 
 * Solution 1:
 *   - do not ask to install a package providing sip3-dev
```
I've seen in the python3-pyqt5 recipe that there are dependencies to both sip-native and sip3-native. From the sip3 recipe, i've tried to remove this portion of code :     
```
# we are inrterested in python3 modules only - for other stuff build sip
rm -rf ${D}/${includedir}
rm -rf ${D}/${bindir}
```
Which drove me to another error saying that there is a conflict between sip and sip3, and bitbake won't install the same files twice at the same place. 

I'm stuck on that for a moment now, any help would be appreciated ! :-)

schnitzeltony, 2018-08-03:
I have python3-pyqt5 in my image and it depends on

```
"python3-pyqt5" -> "libc6"
"python3-pyqt5" -> "libgcc1"
"python3-pyqt5" -> "libqt5svg5"
"python3-pyqt5" -> "libstdc++6"
"python3-pyqt5" -> "python3-core"
"python3-pyqt5" -> "python3-sip"
"python3-pyqt5" -> "qtbase"
```
The question is: What causes your image to contain sip3-dev? To find out: enable buildhistory. That creates a file depends.dot. Search there for sip3-dev (and let me know :)

willdlcre, 2018-08-03:
Hi Andreas, thanks for your answer!
I've checked a few output files and looked for the occurences of 'sip'
Here is what I found :

* From installed-packages.txt:

python3-sip_4.19.2-r0_cortexa9hf-neon.ipk

*From depends.dot :

"python3-pyqt5" -> "libc6"
"python3-pyqt5" -> "libgcc1"
"python3-pyqt5" -> "libqt5svg5"
"python3-pyqt5" -> "libstdc++6"
"python3-pyqt5" -> "python3-core"
"python3-pyqt5" -> "python3-sip"
"python3-pyqt5" -> "qtbase"
"python3-pyqt5" -> "qtwebkit"

There is no other occurence of 'sip' in depends.dot

From the /build/tmp/deploy/ipk/cortexa9 directory :

./sip_4.19.2-r0_cortexa9hf-neon.ipk
./python-sip_4.19.2-r0_cortexa9hf-neon.ipk
./sip3-dbg_4.19.2-r0_cortexa9hf-neon.ipk
./sip3-dev_4.19.2-r0_cortexa9hf-neon.ipk
./sip-dbg_4.19.2-r0_cortexa9hf-neon.ipk
./sip-dev_4.19.2-r0_cortexa9hf-neon.ipk

From  sip3_4.19.2-pyqt5.bb
DEPENDS += "sip sip-native sip3-native qtbase python3"
DEPENDS_append_class-target = "qtsvg qtwebkit"

FILES_sip3 = "${libdir}/${PYTHON_DIR}/site-packages/"
FILES_${PN}-dbg += "${libdir}/${PYTHON_DIR}/site-packages/.debug"

From python3_pyqt5.bb:
RDEPENDS_${PN}_append_class-target = " python3-core python3-sip"

It seems that Bitbake always create a -dev package (same for -doc,
-staticdev, -dbg etc...) but I don't understand why sometimes those
packages are installed on the image, and sometimes not.
I don't really understand how the PACKAGE/FILES system work anyway...

Hope those information can help..

schnitzeltony, 2018-08-03:
For your understanding:
dev packages are created for each recipe. By default dev packages require base package being installed/part of the image.
For your case: sip3-dev requires sip3. Since sip3 is empty and recipe does not contain ALLOW_EMPTY_${PN} = "1", do_rootfs fails.

So the problem is sip3-dev. **What in your environment causes sip3-dev being added to your image**?

* What does depends.dot say for sip3-dev?
* Do you have other -dev files in your image?

Maybe you drop your depends.dot somewhere so I can take a look...

schnitzeltony, 2018-08-04:
For whatever reason your image contains sip3-devel (let's ignore)- the issue you reported caused me to look into and unveiled that python3-pyqt5 was built in the wrong way: It used sip (for python V2) to create bindings which is not a good idea when using python3. 
That (and the problem causing trouble for you) should be fixed. Please pull latest master and let me know.

schnitzeltony, 2018-08-10:
Can we close this?