schnitzeltony/meta-qt5-extra #54: Startkde gets in crashloop

megastallman, 2018-08-18:
Hi @schnitzeltony !

After running startkde I get:
```KActivities: FATAL ERROR: Failed to contact the activity manager daemon
KActivities: FATAL ERROR: Failed to contact the activity manager daemon
KActivities: FATAL ERROR: Failed to contact the activity manager daemon
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 12396, resource id:  4194309, major code: 18 (Change Property), minor code: 0
```
crashlooping.

When I run Xfce4 and try "kactivitymanagerd start" - it says no SQLite driver found, even if add sqlite3 to my packages.

Thanks  in advance!

schnitzeltony, 2018-08-18:
Hi,

Last time I tried kde plasma, it worked without issues - more or less. Currently kwallet fails to compile here so I cannot test. Can you please drop me:

* Your build configuration
* kactivitymanagerd's log.do_configure / log.do_compile

With that I can see what's different at your builds and which changes might fix.

schnitzeltony, 2018-08-18:
A coffee later: Try da916a2396bd912d1cb2f27f3e009a5181dc2767

megastallman, 2018-08-22:
Hi @schnitzeltony !
Thanks much! Now I have a working plasma desktop!!!! At last!
I've been building a kde-world metapackage, but without some dependencies I've excluded:
- kde-games - I'm not this guy.
- minuet - Timidity is broken, and that is the only package requiring it
- recipes-kde/plasma/untier/oxygen-fonts/oxygen-fonts.bb points to a broken link, so I've added
SRC_URI = "https://ftp.osuosl.org/pub/blfs/conglomeration/oxygen-fonts/oxygen-fonts-5.4.3.tar.xz" there.

closed 2018-08-22 by megastallman

schnitzeltony, 2018-08-22:
Hi @megastallman, 

> Thanks much! Now I have a working plasma desktop!!!!

Cool - happy users makes me happy :)

> minuet - Timidity is broken, and that is the only package requiring it

What do you mean by broken? Does timidity not build or is it some run issue?

> recipes-kde/plasma/untier/oxygen-fonts/oxygen-fonts.bb points to a broken link...

Will check that

Thanks for feedback

megastallman, 2018-08-22:
Timidity didn't last week. I think I'll try it again to get logs.