schnitzeltony/meta-qt5-extra #63: plasma-mobile build error:

muddessir, 2018-12-04:
I came across this build error, would you please advice my anything to solve?
**summary:** `ERROR: kdeclarative-5.49.0-r0 do_compile: Function failed: do_compile (log file is located at /home/deb/build-plasma/tmp/work/core2-64-poky-linux/kdeclarative/5.49.0-r0/temp/log.do_compile.11846)
`
full logs: https://paste.ee/p/hKCP2

schnitzeltony, 2018-12-06:
Hope to find some time for this next week...

muddessir, 2018-12-31:
mr. schnitzeltony
did you forget us. we missed you =)

schnitzeltony, 2019-01-01:
Oh yes it slipped out of view silently - sorry...

This issue should be fixed by:

master: https://github.com/schnitzeltony/meta-qt5-extra/commit/abc953bc89441f43fedb54cdf3f717bf2c544470
thud: https://github.com/schnitzeltony/meta-qt5-extra/commit/f808a070528e707bdf20a662bde4a0f378a39ad7
sumo: https://github.com/schnitzeltony/meta-qt5-extra/commit/3f0741da9fce9c60822ee29d85d63c507ad2111b

If it does not fix - please reopen this one.

closed 2019-01-01 by schnitzeltony