schnitzeltony/meta-qt5-extra #67: liri-eglfs compile error on thud

NeenaReddy, 2019-03-21:
I am getting the compile error when building liri-eglfs on thud.

/liri/bld-liri/tmp/work/aarch64-poky-linux/liri-eglfs/0.0.0+gitAUTOINC+abe530108d-r0/git/src/plugins/platforms/eglfs/deviceintegration/eglfs_kms/qeglfskmsgbmscreen.cpp:375:46: **error: 'QOpenGLFunctions_3_0' was not declared in this scope**
     auto funcs = glContext->versionFunctions<QOpenGLFunctions_3_0>();
                                              ^~~~~~~~~~~~~~~~~~~~
/liri/bld-liri/tmp/work/aarch64-poky-linux/liri-eglfs/0.0.0+gitAUTOINC+abe530108d-r0/git/src/plugins/platforms/eglfs/deviceintegration/eglfs_kms/qeglfskmsgbmscreen.cpp:375:46: note: suggested alternative: 'QOpenGLFunctions'
     auto funcs = glContext->versionFunctions<QOpenGLFunctions_3_0>();
                                              ^~~~~~~~~~~~~~~~~~~~
                                              QOpenGLFunctions
/liri/bld-liri/tmp/work/aarch64-poky-linux/liri-eglfs/0.0.0+gitAUTOINC+abe530108d-r0/git/src/plugins/platforms/eglfs/deviceintegration/eglfs_kms/qeglfskmsgbmscreen.cpp:375:68: **error: no matching function for call to 'QOpenGLContext::versionFunctions<<expression error> >()'**
     auto funcs = glContext->versionFunctions<QOpenGLFunctions_3_0>();
                                                                    ^

schnitzeltony, 2019-03-21:
What does

`bitbake -e qtbase | grep ′PACKAGECONFIG`

say?

NeenaReddy, 2019-03-21:
#     "${@bb.utils.contains('PACKAGECONFIG', 'examples', 'QT_BUILD_PARTS+=examples', '', d)}"
#     "${@bb.utils.contains('PACKAGECONFIG', 'tests', 'QT_BUILD_PARTS+=tests', '', d)}"
#     "${@bb.utils.contains('PACKAGECONFIG', 'qtquickcompiler', 'CONFIG+=qtquickcompiler', '', d)}"
#   " ${@bb.utils.contains('PACKAGECONFIG', 'examples', 'QT_BUILD_PARTS+=examples', '', d)} ${@bb.utils.contains('PACKAGECONFIG', 'tests', 'QT_BUILD_PARTS+=tests', '', d)} ${@bb.utils.contains('PACKAGECONFIG', 'qtquickcompiler', 'CONFIG+=qtquickcompiler', '', d)}"
# $PACKAGECONFIG [69 operations]
#     "     ${PACKAGECONFIG_RELEASE}     ${PACKAGECONFIG_DEFAULT}     ${PACKAGECONFIG_OPENSSL}     ${PACKAGECONFIG_GL}     ${PACKAGECONFIG_FB}     ${PACKAGECONFIG_X11}     ${PACKAGECONFIG_KDE}     ${PACKAGECONFIG_FONTS}     ${PACKAGECONFIG_SYSTEM}     ${PACKAGECONFIG_DISTRO} "
#     "${PACKAGECONFIG_WAYLAND}"
#   "     ${PACKAGECONFIG_RELEASE}     ${PACKAGECONFIG_DEFAULT}     ${PACKAGECONFIG_OPENSSL}     ${PACKAGECONFIG_GL}     ${PACKAGECONFIG_FB}     ${PACKAGECONFIG_X11}     ${PACKAGECONFIG_KDE}     ${PACKAGECONFIG_FONTS}     ${PACKAGECONFIG_SYSTEM}     ${PACKAGECONFIG_DISTRO}  ${PACKAGECONFIG_WAYLAND} accessibility examples"
PACKAGECONFIG="     release     dbus udev evdev widgets tools libs freetype tests                openssl     gles2 gbm kms          xcb xinput2 glib xkb xkbcommon-evdev          fontconfig     jpeg libpng zlib           sm     accessibility     sql-sqlite     cups     libinput   xkbcommon-evdev accessibility examples"
# $PACKAGECONFIG_CONFARGS
PACKAGECONFIG_CONFARGS=" -accessibility -cups -dbus -no-directfb -no-eglfs -evdev -make examples -compile-examples -fontconfig -system-freetype -gbm -no-feature-getentropy -no-gif -opengl es2 -glib -no-gtk -qt-harfbuzz -no-ico -no-iconv -no-icu -no-journald -system-libjpeg -kms -libinput -system-libpng -no-libproxy -make libs -no-linuxfb -no-mtdev -openssl -qt-pcre -release -sm -no-sql-db2 -no-sql-ibase -no-sql-mysql -no-sql-oci -no-sql-odbc -no-sql-psql -sql-sqlite -system-sqlite -no-sql-sqlite2 -no-sql-tds -shared -make tests -make tools -no-tslib -libudev -widgets -xcb -xcb-xlib -system-xcb -xinput2 -xkb -xkbcommon-evdev -system-zlib"
# $PACKAGECONFIG_DEFAULT
PACKAGECONFIG_DEFAULT="dbus udev evdev widgets tools libs freetype tests           "
# $PACKAGECONFIG_DISTRO [2 operations]
PACKAGECONFIG_DISTRO="      sm     accessibility     sql-sqlite     cups     libinput "
# $PACKAGECONFIG_FB
PACKAGECONFIG_FB=""
# $PACKAGECONFIG_FONTS [2 operations]
PACKAGECONFIG_FONTS="fontconfig"
# $PACKAGECONFIG_GL [5 operations]
PACKAGECONFIG_GL="gles2 gbm kms"
# $PACKAGECONFIG_GL_imxgpu2d
PACKAGECONFIG_GL_imxgpu2d="gl"
# $PACKAGECONFIG_GL_imxgpu3d
PACKAGECONFIG_GL_imxgpu3d="gles2"
# $PACKAGECONFIG_GL_imxpxp
PACKAGECONFIG_GL_imxpxp="gles2"
# $PACKAGECONFIG_KDE
PACKAGECONFIG_KDE=""
# $PACKAGECONFIG_OPENSSL
PACKAGECONFIG_OPENSSL="openssl"
# $PACKAGECONFIG_RELEASE
PACKAGECONFIG_RELEASE="release"
# $PACKAGECONFIG_SYSTEM
PACKAGECONFIG_SYSTEM="jpeg libpng zlib"
# $PACKAGECONFIG_WAYLAND
PACKAGECONFIG_WAYLAND="xkbcommon-evdev"
# $PACKAGECONFIG_X11
PACKAGECONFIG_X11="xcb xinput2 glib xkb xkbcommon-evdev"
#     "     ${QT_CONFIG_FLAGS_GOLD}     -shared     -silent     -no-pch     -no-rpath     -pkg-config     ${PACKAGECONFIG_CONFARGS} "
#   "      ${QT_CONFIG_FLAGS_GOLD}     -shared     -silent     -no-pch     -no-rpath     -pkg-config     ${PACKAGECONFIG_CONFARGS}  ${QT_CONFIG_FLAGS_APPEND}"
    # Handle PACKAGECONFIG
    # PACKAGECONFIG ??= "<default options>"
    # PACKAGECONFIG[foo] = "--enable-foo,--disable-foo,foo_depends,foo_runtime_depends,foo_runtime_recommends"
    pkgconfigflags = d.getVarFlags("PACKAGECONFIG") or {}
        pkgconfig = (d.getVar('PACKAGECONFIG') or "").split()
                bb.error("%s: PACKAGECONFIG[%s] Only enable,disable,depend,rdepend,rrecommend can be specified!"
        appendVar('PACKAGECONFIG_CONFARGS', extraconf)
    # Check invalid PACKAGECONFIG
    pkgconfig = (d.getVar("PACKAGECONFIG") or "").split()
        pkgconfigflags = d.getVarFlags("PACKAGECONFIG") or {}
                error_msg = "%s: invalid PACKAGECONFIG: %s" % (pn, pconfig)
                        error_msg = "%s rdepends on %s, but it isn't a build dependency, missing %s in DEPENDS or PACKAGECONFIG?" % (pkg, rdepend, rdep_data['PN'])

NeenaReddy, 2019-03-21:
[packageconfig.txt](https://github.com/schnitzeltony/meta-qt5-extra/files/2993660/packageconfig.txt)

schnitzeltony, 2019-03-21:
Ahh Freescale/NXP - what machine exactly?

Just what I do usually / did recently

* my reference machine is Raspi3 with open source graphics stack (VC4 / mesa)
* as far as I can remember I have built/run LIRI for imx6 (Variscite-SOM - see meta-freescale-3rd-party) with open source graphics stack (entaviv/mesa) - maybe I should try another time
* my qtbase is build with PACKAGECONFIG_GL = "gl'  for two reasons: Some projects don't build with qtbase configured for gles and I could not find any performance regression yet.

So let me guess: You use vivante closed source (this either X11 or Wayland thingy...) - or do you think you can give mainline-kernel / etnaviv / qtbase for full OpenGL a try?

schnitzeltony, 2019-03-21:
One bit off topic note: be aware that LIRI is WIP and has issues. Just to avoid disappointments...

NeenaReddy, 2019-03-21:
I am using imx8qmmek machine.

I built Liri before for the same machine when all the yocto recipes in meta-qt5-extra pointed to old commits of Liri source. ( I think this is because Liri adapted qbs build system in between). I started building it again after i learnt that Liri moved to cmake but came across this build break.

Yes, we use vivante closed source. I am able to see those openGL functions in the sysroots of Liri-eglfs .
 I will try with the mainline-kernel but not sure if it will work.

schnitzeltony, 2019-03-21:
> I built Liri before for the same machine when all the yocto recipes in meta-qt5-extra pointed to old commits of Liri source. ( I think this is because Liri adapted qbs build system in between).

Yes I did not get QBS to fly - but that is ages ago...

schnitzeltony, 2019-03-22:
Sleeping this over made me remember: had trouble at the same code area not during build but at runtime [1].

There is some frame-recorder (enabled by default upstream - hmm)  which uses OpenGL3 functions. Up to now this was was just disabled - now I removed the code completely [2].
Be aware that I rebased thud branch to master to the point just before music stuff was splitted out to meta-musicians. I did that to avoid backport dance and to get best out of thud branch.

Long talk short: pulling meta-qt5-extra/thud should fix your issue.

[1] https://github.com/lirios/lirios/issues/61
[2] https://github.com/schnitzeltony/meta-qt5-extra/commit/25359434082002fe1a8aebc505b46a097d052b79

NeenaReddy, 2019-03-22:
Thank you! Pulling meta-qt5-extra/thud  fixed my build break.

closed 2019-03-22 by schnitzeltony