schnitzeltony/meta-qt5-extra #68: System icons not found

alexgg, 2019-04-10:
Hi,
I am using the thud branch and I get a bunch of errors after launching "liri-session":
`
file:///usr/lib/qml/Fluid/Controls/Icon.qml:47:5: QML Image: Failed to get image from provider: image://fluidicontheme/io.liri.Fluid.Demo
`
`
file:///usr/lib/qml/Fluid/Controls/Icon.qml:47:5: QML Image: Failed to get image from provider: image://fluidicontheme/io.liri.Browser
`
As a result no system icons are displayed. I can see some icons are installed under /usr/share/icons/hicolor:

`
find -name io.liri.Browser*
`
`./usr/share/applications/io.liri.Browser.desktop`
`./usr/share/icons/hicolor/64x64/apps/io.liri.Browser.png`
`./usr/share/icons/hicolor/scalable/apps/io.liri.Browser.svg`
`./usr/share/icons/hicolor/512x512/apps/io.liri.Browser.png`
`./usr/share/icons/hicolor/192x192/apps/io.liri.Browser.png`
`./usr/share/icons/hicolor/128x128/apps/io.liri.Browser.png`
`./usr/share/icons/hicolor/16x16/apps/io.liri.Browser.png`
`./usr/share/icons/hicolor/256x256/apps/io.liri.Browser.png`
`./usr/share/icons/hicolor/32x32/apps/io.liri.Browser.png`

Any hints?

Thanks,

schnitzeltony, 2019-04-10:
No hints yet but was already on my never ending TODO...

schnitzeltony, 2020-10-14:
As far as I can remember there was an enhancement - no full fix - on this. However: have not seen much activity on Liri so close this one. Feel free to re-open.

closed 2020-10-14 by schnitzeltony