schnitzeltony/meta-qt5-extra #71: Warrior support

schultzern, 2019-06-05:
Are you planning to support the warrior release :) ?

schnitzeltony, 2019-06-05:
Yes but have a different timing than major layers: in my layers I create stable branches as late as possible - means when modifications make layer incompatible to latest stable branch (this time warrior). I do this to have the latests updates included in last release and to reduce my maintenance efforts (have 5 layers - meta-raspi-light excluded).

FWIW: My productive builds are based on oe-core warrior (oe-core master is just build tested from time to time)

If you know of issues or have strong reasons that make it necessary to create warrior soon - please let me know.

schultzern, 2019-06-06:
Thanks for the info! I don't have any specific reason for updating right now other than wanting to go to the "latest" when I do. I'll keep checking in. 

Thank you for maintaining this layer!

schnitzeltony, 2019-08-12:
Just created warrior branch

closed 2019-08-12 by schnitzeltony