schnitzeltony/meta-qt5-extra #73: Errors with OpenSTLinux-x11 Distro

vbursucianu, 2019-10-12:
Steps to reproduce:
`$ mkdir openstlinux-4.19-thud-mp1-19-10-09 `
`$ cd openstlinux-4.19-thud-mp1-19-10-09`
`$ repo init -u https://github.com/STMicroelectronics/oe-manifest.git -b refs/tags/openstlinux-4.19-thud-mp1-19-10-09`
`$ repo sync`
`$ DISTRO=openstlinux-x11 MACHINE=stm32mp1 source layers/meta-st/scripts/envsetup.sh`
`$ bitbake-layers layerindex-fetch meta-qt5-extra`
`$ bitbake-layers show-layers `

This should look like this::
```
NOTE: Starting bitbake server...
layer                 path                                      priority
==========================================================================
meta-python           /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-python  7
meta-oe               /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-oe  6
meta-oe               /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-oe  6
meta-gnome            /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-gnome  7
meta-xfce             /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-xfce  7
meta-initramfs        /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-initramfs  8
meta-multimedia       /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-multimedia  6
meta-networking       /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-networking  5
meta-webserver        /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-webserver  6
meta-filesystems      /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-filesystems  6
meta-perl             /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-perl  6
meta-python           /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-python  7
meta-st-stm32mp       /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-st/meta-st-stm32mp  6
meta-qt5              /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-qt5  7
meta-st-openstlinux   /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-st/meta-st-openstlinux  5
meta                  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta  5
meta-qt5-extra        /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra  20
```
Make sure the repos are checked out with <refs/tags/openstlinux-4.19-thud-mp1-19-10-09> if possible.
Branches:
```
project layers/openembedded-core/meta-qt5-extra/   branch origin/master
project layers/meta-qt5/                        branch github/master
project layers/meta-st/meta-st-openstlinux/     branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
project layers/meta-openembedded/               branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
project layers/meta-st/meta-st-stm32mp/         branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
project layers/meta-st/scripts/                 branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
project layers/openembedded-core/               branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
project layers/meta-timesys/                    branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
project layers/openembedded-core/bitbake/       branch refs/tags/openstlinux-4.19-thud-mp1-19-10-09
```
I configured the plasma-mobile into <VIRTUAL-RUNTIME_graphical_init_manager> 
`$ vi ../layers/meta-st/meta-st-openstlinux/conf/distro/openstlinux-x11.conf`
```
require include/st-default-distro-rules.inc
require include/st-default-distro-providers.inc
require include/openstlinux.inc

DISTRO = "openstlinux-x11"
DISTRO_NAME = "ST OpenSTLinux - X11 - (A Yocto Project Based Distro)"
TARGET_VENDOR = "-openstlinux_x11"
SDK_VENDOR = "-openstlinux_x11_sdk"

DISTRO_FEATURES_append = " opengl"
DISTRO_FEATURES_append = " gplv3 qt5 kde"
VIRTUAL-RUNTIME_graphical_init_manager = "plasma-mobile"
```
I launch my build with:
`$ bitbake -k st-example-image-x11`

I endup having 7 errors witch 6 are from the layer meta-qt-extra that is compatible with thud:
```
Summary: 7 tasks failed:
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/plasma/tier1/kdecoration/kdecoration.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier1/kplotting/kplotting.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier1/kitemviews/kitemviews.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier1/sonnet/sonnet.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/plasma/untier/libkscreen/libkscreen.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-qt5/recipes-qt/qt5/qtwayland_git.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier2/kfilemetadata/kfilemetadata.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 8 ERROR messages shown, returning a non-zero exit code.
```
Some details:
1) kdecoration.bb:do_configure log shows:
```
|   The following configuration files were considered but not accepted:
|
|     /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/kdecoration/5.16.5-r0/recipe-sysroot/usr/lib/cmake/Qt5/Qt5Config.cmake, version: 5.11.3
|     /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/kdecoration/5.16.5-r0/recipe-sysroot-native/usr/lib/cmake/Qt5/Qt5Config.cmake, version: 5.11.3
```
2) kplotting.bb  log shows:
```
| -- The following REQUIRED packages have not been found:
|
|  * Qt5UiPlugin
|    Required to build Qt Designer plugins
```
3) sonnet.bb  log shows:
```
| -- The following REQUIRED packages have not been found:
|
|  * Qt5UiPlugin
|    Required to build Qt Designer plugins
```
4) libkscreen.bb
```
|   The following configuration files were considered but not accepted:
|
|     /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/libkscreen/5.16.5-r0/recipe-sysroot/usr/lib/cmake/Qt5/Qt5Config.cmake, version: 5.11.3
|     /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/libkscreen/5.16.5-r0/recipe-sysroot-native/usr/lib/cmake/Qt5/Qt5Config.cmake, version: 5.11.3
```
5) qtwayland_git.bb:
```
| ERROR: Feature 'xcomposite-glx' was enabled, but the pre-condition 'features.wayland-client && features.opengl && !features.opengles2 && features.xlib && libs.xcomposite && libs.glx' failed.
|
| ERROR: Feature 'xcomposite-glx' was enabled, but the pre-condition 'features.wayland-server && features.opengl && !features.opengles2 && libs.xcomposite && libs.glx' failed.
```
6) kfilemetadata :
```
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxx11-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-openstlinux_x11-linux-gnueabi/kfilemetadata/5.62.0-r0/temp/log.do_compile.11046)
```

Anyone can help me? Thanks

schnitzeltony, 2019-10-12:
Have to ask for more information:

1. Exact revisions of your layer (branch/commits). Please drop the build configuration seen on console when running 'bitbake -k st-example-image-x11'
2. Your local.conf (you can upload it here by renaming it local.conf -> local.conf.log)
3. Same for bblayers.conf

After that I will try to reproduce your set-up steps and see what I get...

schnitzeltony, 2019-10-12:
Looked in your logs again: Please try thud branch for meta-qt5-extra

I know your layer.conf in master says it is still compatible to thud but that does not seem to be true any more so I corrected that [1]

[1] https://github.com/schnitzeltony/meta-qt5-extra/commit/9c6b7ee064ee0241ad3aef6bc8c0be5adb6eed7c

vbursucianu, 2019-10-12:
Hi, thanks for looking into this issue,
Here is my configuration seen on my console:
```
 $ bitbake -k st-example-image-x11                  NOTE: Started PRServer with DBfile: /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/build-openstlinuxx11-stm32mp1/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 37855, PID: 29683
Parsing recipes: 100% |#############| Time: 0:00:42Parsing of 2833 .bb files complete (0 cached, 2833 parsed). 3876 targets, 83 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-openstlinux_x11-linux-gnueabi"
MACHINE              = "stm32mp1"
DISTRO               = "openstlinux-x11"
DISTRO_VERSION       = "2.6-snapshot-20191012"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
DISTRO_CODENAME      = "thud"
ACCEPT_EULA_stm32mp1 = "1"
GCCVERSION           = "8.%"
PREFERRED_PROVIDER_virtual/kernel = "linux-stm32mp"meta-python
meta-oe
meta-oe
meta-gnome
meta-xfce
meta-initramfs
meta-multimedia
meta-networking
meta-webserver
meta-filesystems
meta-perl
meta-python          = "refs/tags/openstlinux-4.19-thud-mp1-19-10-09:9b3b907f30b0d5b92d58c7e68289184fda733d3e"
meta-st-stm32mp      = "heads/refs/tags/openstlinux-4.19-thud-mp1-19-10-09:d8cbac759e1275b1a27d4ba38b64a0d83d0e8c9f"
meta-qt5             = "heads/github/thud:1520d5b2b2beec5e1c3209d3178219e93ef08bca"
meta-st-openstlinux  = "heads/refs/tags/openstlinux-4.19-thud-mp1-19-10-09:ee276cb390d7dd1175d386ba2bfd384c1b206808"
meta                 = "refs/tags/openstlinux-4.19-thud-mp1-19-10-09:d3d3f443039b03f1200a14bfe99f985592632018"
meta-qt5-extra       = "heads/origin/thud:09ae5e345a5e3559e6237b93edf956cfd52e4dcf"
```
The build still fails even on the thud branch from the layer meta-qt5 and meta-qt5-extra.
Looks like the errors are the same:
```
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user                     # to the system might want to change but pretty much any configuration option can                     # be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file                     # but new users likely won't need any of them initially.
#                                                  #                                                  # This file is your local configuration file and is where all local user settings                     #                                                  # This file is your local configuration file and is where all local user settings                     # are placed. The comments in this file give some guide to the options a new user                     # to the system might want to change but pretty much any configuration option can                     # be set in this file. More adventurous users can look at local.conf.extended                         # which contains other examples of configuration which can be placed in this file                     # but new users likely won't need any of them initially.                                              #                                                  # Lines starting with the '#' character are commented out and in some cases the                       # default values are provided as comments to show people example syntax. Enabling                     conf/local.conf                  1,1            Top[ns@yocto build-openstlinuxx11-stm32mp1]           $ cat conf/local.conf
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
[... 324 lines cut ...]
    ${OEROOT}/layers/meta-openembedded/meta-filesystems' \
    if os.path.isfile('${OEROOT}/layers/meta-security/conf/layer.conf') else ''} \
    "

# specific to framework
FRAMEWORKLAYERS += "${@'${OEROOT}/layers/meta-st/meta-st-openstlinux' if os.path.isfile('${OEROOT}/layers/meta-st/meta-st-openstlinux/conf/layer.conf') else ''}"

# add BSP layer
BBLAYERS += " \
  ${BASELAYERS} \
  ${BSPLAYER} \
  ${ADDONSLAYERS} \
  ${FRAMEWORKLAYERS} \
  ${@'${OPENEMBEDDED}' \
  if \
  os.path.isfile('${OEROOT}/layers/openembedded-core/meta/conf/layer.conf') \
  else \
  '${POKY}'} \
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra \
  "
# BSP dependencies"
BBLAYERS =+ "/home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-oe"
BBLAYERS =+ "/home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-openembedded/meta-python"
[ns@yocto build-openstlinuxx11-stm32mp1]
```

vbursucianu, 2019-10-12:
The build failed with those errors when using the thud branch for meta-qt5 and meta-qt5-extra:
```
NOTE: Tasks Summary: Attempted 11464 tasks of which 11457 didn't need to be rerun and 7 failed.
NOTE: Writing buildhistory
Summary: 7 tasks failed:
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier1/kitemviews/kitemviews.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier1/sonnet/sonnet.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier2/kfilemetadata/kfilemetadata.bb:do_compile
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/kf5/tier1/kplotting/kplotting.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/plasma/tier1/kdecoration/kdecoration.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/openembedded-core/meta-qt5-extra/recipes-kde/plasma/untier/libkscreen/libkscreen.bb:do_configure
  /home/ns/Documents/stm32/distribution-package/openstlinux-4.19-thud-mp1-19-10-09/layers/meta-qt5/recipes-qt/qt5/qtwayland_git.bb:do_configure
Summary: There were 8 ERROR messages shown, returning a non-zero exit code.
[ns@yocto build-openstlinuxx11-stm32mp1]
```
Looks like the same issues

vbursucianu, 2019-10-27:
Hi schnitzeltony,
Any success on solving this issue? Thanks :)

schnitzeltony, 2019-10-27:
Sorry but I do have the resources currently:

* am working on gnome patches for meta-openembedded
* have one build machine at home only

One hint maybe helpful:
The last error is meta-qt5*s qtwayland. Am not sure but do you have wayland in your DISTRO_FEATURES?

To check run
bitbake -e | grep ^DISTRO_FEATURES

Maybe you can post the result here.

schnitzeltony, 2020-10-14:
No activity for long time - close it

closed 2020-10-14 by schnitzeltony