schnitzeltony/meta-qt5-extra #78: Cannot bring up liri desktop on zeus using yocto image

NeenaReddy, 2020-03-05:
My liri yocto build was successful on zeus but when i flash my sdcard and boot the image, LiriDesktop doesn't come up.  Do i need to make any other additional changes to bring it up other than building liri recipes ?

schnitzeltony, 2020-03-05:
Have just tested with master - it is running - with some help:

Have sddm as display manager in my images and selecting LIRI for the first time does not work but with session autostart enabled and LIRI selected LIRI works fine. Might be some sddm pam issue because sddm login and sddm autologin have different pam configs.

NeenaReddy, 2020-03-05:
Hi Andreas,

Thank you for your reply. I never used sddm and i am not aware of it.
Could you please tell me what files you have changed to get liri working.

NeenaReddy, 2020-03-06:
I built sddm display manager and on boot up it is directly executing X sessions, although i enabled wayland in sddm.conf

 sddm[827]: Initializing...
 sddm[827]: Starting...
 sddm[827]: Logind interface found
sddm[827]: Adding new display on vt 1 ...
sddm[827]: Loading theme configuration from ""
sddm[827]: Display server starting...
sddm[827]: Running: /usr/bin/X -nolisten tcp -auth /var/run/sddm/{5015c2d7-415a-4fb2-84fd-f5da406aff92} -background none -noreset -displayfd 18 -seat seat0 vt1
sddm[827]: Failed to start display server process.
sddm[827]: Display server failed to start. Exiting
systemd[1]: sddm.service: Main process exited

**Here is my /etc/sddm.conf**

[Autologin]

Session=liri.desktop
User=
Relogin=true

[Wayland]

SessionCommand=/usr/share/sddm/scripts/wayland-session
SessionDir=/usr/share/wayland-sessions
SessionLogFile=/var/log/wayland-session.log

I am i missing something ?

NeenaReddy, 2020-03-09:
@schnitzeltony

Is sddm supported only for x11 ? If not, how do i configure it for wayland ?

schnitzeltony, 2020-03-09:
Sorry it was my fault: Forgot that you have either x11 or wayland on imx blob. sddm greeter starts x11 and efforts to start it for wayland were never applied.

Unfortunately I have zero time to take care - am under pressure... As soon as this changes I will try to reproduce your case

NeenaReddy, 2020-03-18:
@schnitzeltony : Sorry to bug you again. 
I made some progress on this and now i am getting PAM authentication failure when i start liri with SDDM.  In your previous comment, you mentioned that you ran into this issue when you started liri for the first time
Could you please tell me what changes you made to get it working and also can you share your sddm.conf file ?

schnitzeltony, 2020-03-18:
> I made some progress on this and now i am getting PAM authentication failure when i start liri with SDDM.

Could you share some log - I am still interested in a solution

>  Could you please tell me what changes you made to get it working and also can you share your sddm.conf file ?

Have created a distro-layer called meta-mortsgna. It creates a default user 'morona'. To make that user autologin-capable it stores a simple sddm-conf-snippet in /etc/sddm.conf.d containing:

`[Autologin]
User=morona`

See [1] for further details

With this 
* try a sddm-login -> that fails
* power off / power on -> autologin works for me

[1] https://github.com/schnitzeltony/meta-mortsgna/blob/master/recipes-defaultconfig/users/sddm-morona-autologin.bb

schnitzeltony, 2020-10-14:
No activity for long time - will close it. If it is still an issue bugging you, feel free to re-open

closed 2020-10-14 by schnitzeltony