schnitzeltony/meta-qt5-extra #79: Layer bbappend for qtbase-native causing build failure

kekiefer, 2020-06-02:
With current master branches, I have the following build failure in qtbase-native due to the bbappend in this layer. Note that I am using GCC 8 in my builds. I haven't really studied the change, but it's not clear why the patch needs to remove the namespace on `Qt::endl`.

```
| /home/kkiefer/code/yocto/poky/build-ccu3/tmp/work/x86_64-linux/qtbase-native/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/src/tools/qdbusxml2cpp/qdbusxml2cpp.cpp: In function ‘QTextStream& writeHeader(QTextStream&, bool)’:
| /home/kkiefer/code/yocto/poky/build-ccu3/tmp/work/x86_64-linux/qtbase-native/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/src/tools/qdbusxml2cpp/qdbusxml2cpp.cpp:169:52: error: ‘endl’ was not declared in this scope
|         << " * Command line was: " << strCmdLine << endl
|                                                     ^~~~
| /home/kkiefer/code/yocto/poky/build-ccu3/tmp/work/x86_64-linux/qtbase-native/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/src/tools/qdbusxml2cpp/qdbusxml2cpp.cpp:169:52: note: suggested alternatives:
| In file included from /usr/include/c++/7/iterator:64:0,
|                  from ../../../include/QtCore/../../../git/src/corelib/text/qbytearray.h:53,
|                  from ../../../include/QtCore/qbytearray.h:1,
|                  from /home/kkiefer/code/yocto/poky/build-ccu3/tmp/work/x86_64-linux/qtbase-native/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/src/tools/qdbusxml2cpp/qdbusxml2cpp.cpp:29:
| /usr/include/c++/7/ostream:590:5: note:   ‘std::endl’
|      endl(basic_ostream<_CharT, _Traits>& __os)
|      ^~~~
| In file included from ../../../include/QtCore/qtextstream.h:1:0,
|                  from ../../../include/QtCore/../../../git/src/corelib/io/qdebug.h:49,
|                  from ../../../include/QtCore/qdebug.h:1,
|                  from /home/kkiefer/code/yocto/poky/build-ccu3/tmp/work/x86_64-linux/qtbase-native/5.15.0+gitAUTOINC+f6fe4bbab7-r0/git/src/tools/qdbusxml2cpp/qdbusxml2cpp.cpp:32:
| ../../../include/QtCore/../../../git/src/corelib/serialization/qtextstream.h:261:28: note:   ‘Qt::endl’
|  Q_CORE_EXPORT QTextStream &endl(QTextStream &s);
|                             ^~~~
| Makefile:473: recipe for target '.obj/qdbusxml2cpp.o' failed
| make[2]: *** [.obj/qdbusxml2cpp.o] Error 1
| make[2]: Leaving directory '/home/kkiefer/code/yocto/poky/build-ccu3/tmp/work/x86_64-linux/qtbase-native/5.15.0+gitAUTOINC+f6fe4bbab7-r0/build/src/tools/qdbusxml2cpp'
| Makefile:418: recipe for target 'sub-qdbusxml2cpp-make_first' failed
| make[1]: *** [sub-qdbusxml2cpp-make_first] Error 2
```

schnitzeltony, 2020-06-02:
Thanks for reporting.

Background: It was introduced because full paths to sysroot were set in source code file of qdbusxml2cpp that lead to ugly runtime errors  in KDE and others. To detect these I created cmake_extra_sanity.bbclass.
Don't remember why I replaced `Qt::endl` by `endl` (maybe it happened accidentally by some patch refresh). This replacement looks wrong.

So I will rework the patch so it sets `Qt::endl`. Unfortunately my images take days to build so you might want to try replacing `endl` by `Qt::endl` in 0002-qdbusxml2cpp-prepare-commandline-string-manipulation.patch and let me know if it fixes your problem.

schnitzeltony, 2020-06-02:
> maybe it happened accidentally by some patch refresh

The culprit was:
ad7f7320df6865bd083e5c744247867fc09da947

schnitzeltony, 2020-06-02:
Looked into some more - it's so long time ago..  The append/patch does nothing else but changing the comment in generated source code to avoid false positives in cmake_extra_sanity.bbclass.

closed 2020-06-02 by schnitzeltony