schnitzeltony/meta-qt5-extra #85: Problems compiling qtbase

yong-ruilin, 2021-05-21:
Problems compiling qtbase：
cd qalgorithms/ && ( test -e Makefile || /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/recipe-sysroot-native/usr/bin/qmake -o Makefile /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/git/tests/auto/corelib/tools/qalgorithms/qalgorithms.pro -qtconf /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/bin/qt.conf ) && make -f Makefile 
{standard input}: Assembler messages:
{standard input}:1309656: Warning: end of file not at end of a line; newline inserted
{standard input}:1310678: Warning: missing operand; zero assumed
linking tst_qmenubar
linking tst_qmdisubwindow
linking tst_qmdiarea
linking tst_qwidget
{standard input}:1310655: Error: invalid operands (*UND* and .gcc_except_table sections) for `-'
linking tst_qplaintextedit
make[5]: Leaving directory '/home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/tests/auto/widgets/widgets/qmenubar'
cd qarraydata/ && ( test -e Makefile || /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/recipe-sysroot-native/usr/bin/qmake -o Makefile /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/git/tests/auto/corelib/tools/qarraydata/qarraydata.pro -qtconf /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/bin/qt.conf ) && make -f Makefile 
make[5]: Leaving directory '/home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/tests/auto/widgets/widgets/qmdisubwindow'
cd qarraydata_strictiterators/ && ( test -e Makefile || /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/recipe-sysroot-native/usr/bin/qmake -o Makefile /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/git/tests/auto/corelib/tools/qarraydata_strictiterators/qarraydata_strictiterators.pro -qtconf /home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/bin/qt.conf ) && make -f Makefile 
mips64el-loongson-linux-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
Makefile:1803: recipe for target '.obj/tst_qstatemachine.o' failed
make[5]: *** [.obj/tst_qstatemachine.o] Error 1
make[5]: Leaving directory '/home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/tests/auto/corelib/statemachine/qstatemachine'
Makefile:72: recipe for target 'sub-qstatemachine-make_first' failed
make[4]: *** [sub-qstatemachine-make_first] Error 2
make[4]: Leaving directory '/home/ubuntu/yocto/poky/build/tmp/work/gs464e-loongson-linux/qtbase/5.15.2+gitAUTOINC+40143c189b-r0/build/tests/auto/corelib/statemachine'
Makefile:284: recipe for target 'sub-statemachine-make_first' failed
make[3]: *** [sub-statemachine-make_first] Error 2

Please ask how to solve this problem。

schnitzeltony, 2021-05-21:
> mips64el-loongson-linux-g++: fatal error: Killed signal terminated program cc1plus

This is most likely caused by running out of memory. You can
* reduce BB_NUMBER_THREADS in local.conf
* reduce PARALLEL_MAKE in local.conf
* add more RAM to your build machine

yong-ruilin, 2021-05-21:
Okay, thanks, I have compiled it now.

closed 2021-05-21 by yong-ruilin