schnitzeltony/meta-qt5-extra #86: KArchiveFile data() returns less data then size()

kpopv, 2021-06-08:
I'm experiencing a bug with KArchive 5.73 in the dunfell branch which I reported to KDE: https://bugs.kde.org/show_bug.cgi?id=437778 . However, it appears this bug isn't triggered when building the same KArchive from source so I'm looking into whether this is something yocto specific. 

Attached is a minimal example to reproduce the issue. Is someone able to confirm whether or not this is an issue?
[tst_KArchiveTest.txt](https://github.com/schnitzeltony/meta-qt5-extra/files/6616468/tst_KArchiveTest.txt)

dev-0x7C6, 2021-06-08:
Maybe libraries that're used by it causing problem? Like you suggested libz, maybe it's worth to try different versions? https://layers.openembedded.org/layerindex/recipe/89425/

Not sure if KArchive is using libz for zip files, but probably is.

kpopv, 2021-06-09:
Thank you for the suggestion, indeed, it seems related to the specific libz we're using:

- libz.so.1.2.11: OK
- libz.so.1.2.11.1-motley: NOT OK

I've filed a bug against that specific version's project: https://github.com/jtkukunas/zlib/issues/33

schnitzeltony, 2021-06-09:
Just for the record: As soon as I can be of support - let me know :parking:

kpopv, 2021-06-21:
As we've verified https://github.com/jtkukunas/zlib/issues/33 is the root cause, this issue can be closed. Thank you for the support!

closed 2021-06-21 by kpopv